groove-x/mqtt_bridge #3: Error when i try send data from mqtt to ros

DBeloglazov, 2017-11-11:
Hello. When I try to send a message from MQTT mqtt_bridge gives me an error. I try to send this command from ubuntu terminal: mosquitto_pub -t echo -m "hello". Where is my error?

Echo param configurations:
  # echo back
  - factory: mqtt_bridge.bridge:RosToMqttBridge
    msg_type: std_msgs.msg:String
    topic_from: /echo
    topic_to: echo
  - factory: mqtt_bridge.bridge:MqttToRosBridge
    msg_type: std_msgs.msg:String
    topic_from: echo
    topic_to: /back

Error mesaage:
[INFO] [1510431743.134914]: MQTT connected
Exception in thread Thread-72:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/usr/local/lib/python2.7/dist-packages/paho/mqtt/client.py", line 2650, in _thread_main
    self.loop_forever(retry_first_connection=True)
  File "/usr/local/lib/python2.7/dist-packages/paho/mqtt/client.py", line 1481, in loop_forever
    rc = self.loop(timeout, max_packets)
  File "/usr/local/lib/python2.7/dist-packages/paho/mqtt/client.py", line 1003, in loop
    rc = self.loop_read(max_packets)
  File "/usr/local/lib/python2.7/dist-packages/paho/mqtt/client.py", line 1284, in loop_read
    rc = self._packet_read()
  File "/usr/local/lib/python2.7/dist-packages/paho/mqtt/client.py", line 1849, in _packet_read
    rc = self._packet_handle()
  File "/usr/local/lib/python2.7/dist-packages/paho/mqtt/client.py", line 2305, in _packet_handle
    return self._handle_publish()
  File "/usr/local/lib/python2.7/dist-packages/paho/mqtt/client.py", line 2500, in _handle_publish
    self._handle_on_message(message)
  File "/usr/local/lib/python2.7/dist-packages/paho/mqtt/client.py", line 2642, in _handle_on_message
    callback(self, self._userdata, message)
  File "/home/denis/fpi_competition/catkin_ws/src/mqtt_bridge/src/mqtt_bridge/bridge.py", line 114, in _callback_mqtt
    ros_msg = self._create_ros_message(mqtt_msg)
  File "/home/denis/fpi_competition/catkin_ws/src/mqtt_bridge/src/mqtt_bridge/bridge.py", line 124, in _create_ros_message
    msg_dict = self._deserialize(mqtt_msg.payload)
  File "msgpack/_unpacker.pyx", line 142, in msgpack._unpacker.unpackb (msgpack/_unpacker.cpp:142)
ExtraData: unpack(b) received extra data.

ledmonster, 2017-11-15:
@DBeloglazov this is because msgpack serializer/deserializer requires binary encoded/decoded data. If you want to try with simple string, please use json instead.

change config file as following:

```
- serializer: msgpack:dumps
- deserializer: msgpack:loads
+ serializer: json:dumps
+ deserializer: json:loads
```
Also, you have to send a string as 'json' like this.

```
mosquitto_pub -t echo -m '"hello"'
```

You have to quote `hello` with double quotation: `"hello"`.

Alkarex, 2017-11-28:
After changing to JSON serializer/deserializer:

@ledmonster The following command fails:
```sh
mosquitto_pub -t echo -m '"hello"'
```
With this error:
```
File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosbridge_library/internal/message_conversion.py", line 258, in _to_object_inst
    raise FieldTypeMismatchException(roottype, stack, rostype, type(msg))
FieldTypeMismatchException: Expected a JSON object for type std_msgs/String but received a <type 'unicode'>
```

@DBeloglazov However, the following seems to work fine:
```sh
mosquitto_pub -t '/test' -m '{"data": "Hello World!"}'
```

It looks like only JSON objects are accepted, not other JSON types such as string.

ledmonster, 2018-01-22:
@Alkarex @DBeloglazov Sorry for replying late. I'll check the issue this weekend.

cihankurt98, 2020-12-02:
Sorry for waking up this thread again, but is this problem solved? Because I can't send a unicode JSON string over the bridge.

ledmonster, 2021-01-01:
Sorry for replying too late.

@Alkarex 's approach is correct. Primitive data should be decorated with `data` key by specification.

@cihankurt98 What kind of data do you send from mqtt client?

closed 2021-10-09 by ledmonster