groove-x/mqtt_bridge #9: Error in: roslaunch mqtt_bridge demo.launch

hesamira, 2018-02-18:
Hi,
I have faced with the following errors.
Please, guide me. 
Thanks.

sami@sami-Vostro-1520:~/mqtt$ roslaunch mqtt_bridge demo.launch
... logging to /home/sami/.ros/log/3a2302a4-147b-11e8-ba87-904ce5b24de1/roslaunch-sami-Vostro-1520-6658.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://sami-Vostro-1520:36279/

SUMMARY
========

PARAMETERS
 * /mqtt_bridge/bridge: [{'topic_from': '...
 * /mqtt_bridge/deserializer: msgpack:loads
 * /mqtt_bridge/mqtt/client/protocol: 4
 * /mqtt_bridge/mqtt/connection/host: localhost
 * /mqtt_bridge/mqtt/connection/keepalive: 60
 * /mqtt_bridge/mqtt/connection/port: 1883
 * /mqtt_bridge/mqtt/private_path: device/001
 * /mqtt_bridge/serializer: msgpack:dumps
 * /rosdistro: indigo
 * /rosversion: 1.11.21

NODES
  /
    mqtt_bridge (mqtt_bridge/mqtt_bridge_node.py)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
running rosparam delete /mqtt_bridge/
process[mqtt_bridge-1]: started with pid [6681]
Traceback (most recent call last):
  File "/home/sami/mqtt/src/mqtt_bridge/scripts/mqtt_bridge_node.py", line 5, in <module>
    from mqtt_bridge.app import mqtt_bridge_node
  File "/home/sami/mqtt/src/mqtt_bridge/src/mqtt_bridge/app.py", line 4, in <module>
    import inject
ImportError: No module named inject
[mqtt_bridge-1] process has died [pid 6681, exit code 1, cmd /home/sami/mqtt/src/mqtt_bridge/scripts/mqtt_bridge_node.py __name:=mqtt_bridge __log:=/home/sami/.ros/log/3a2302a4-147b-11e8-ba87-904ce5b24de1/mqtt_bridge-1.log].
log file: /home/sami/.ros/log/3a2302a4-147b-11e8-ba87-904ce5b24de1/mqtt_bridge-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done

yuma-m, 2018-02-18:
Hi, @hesamira 

I think inject is missing.
You can install it via pip.

`sudo -H pip install inject`

hesamira, 2018-02-18:
I tested that command, but encountered the same error again.

yuma-m, 2018-02-19:
Does the result of command `pip freeze|grep -i inject` is like below?
```
$ pip freeze|grep -i inject
Inject==3.3.2
```

If it is empty, something was wrong with the installation of `inject` module. 

Which do you use python2 or python3?
Please try `sudo -H pip2 -r requirements.txt` or `sudo -H pip3 -r requirements.txt`, to specify which python for inject to be installed.

hesamira, 2018-02-19:
Thanks. My Python version is  2.7.6
I have faced this error, again:

ImportError: No module named paho.mqtt.client

yuma-m, 2018-02-19:
Please try `sudo -H pip2 install bson inject paho-mqtt msgpack-python`.  Mandantory python modules will be installed.
I updated README.md and requirements.txt in PR #10 .

hesamira, 2018-02-20:
Thanks.
I am beginner in python. Please, guide me. 
I have faced with this error: 
socket.error: [Errno 111] Connection refused

yuma-m, 2018-02-21:
`demo.launch` will create a connection to mqtt broker on `localhost:1883`.
Is mqtt broker (like [mosquitto](https://mosquitto.org/)) running on your computer?

hesamira, 2018-02-22:
Unfortunately, no.

yuma-m, 2018-02-22:
If your mosquitto is not running, please start it by `sudo service mosquitto start` or `sudo systemctl start mosquitto.service`.
Then try `demo.launch`.

hesamira, 2018-02-22:
Thanks.  I have tested it. It  was correct.
sami@sami-Vostro-1520:~/mqtt$ sudo service mosquitto start
mosquitto start/running, process 15942

But I have faced this error, again.
sami@sami-Vostro-1520:~/mqtt$ source devel/setup.bash 
sami@sami-Vostro-1520:~/mqtt$ roslaunch mqtt_bridge demo.launch
... logging to /home/sami/.ros/log/eb832402-17dc-11e8-a3ed-904ce5b24de1/roslaunch-sami-Vostro-1520-16402.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://sami-Vostro-1520:40794/

SUMMARY
========

PARAMETERS
 * /mqtt_bridge/bridge: [{'topic_from': '...
 * /mqtt_bridge/deserializer: msgpack:loads
 * /mqtt_bridge/mqtt/client/protocol: 4
 * /mqtt_bridge/mqtt/connection/host: localhost
 * /mqtt_bridge/mqtt/connection/keepalive: 60
 * /mqtt_bridge/mqtt/connection/port: 1883
 * /mqtt_bridge/mqtt/private_path: device/001
 * /mqtt_bridge/serializer: msgpack:dumps
 * /rosdistro: indigo
 * /rosversion: 1.11.21

NODES
  /
    mqtt_bridge (mqtt_bridge/mqtt_bridge_node.py)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
running rosparam delete /mqtt_bridge/
process[mqtt_bridge-1]: started with pid [16425]
Traceback (most recent call last):
  File "/home/sami/mqtt/src/mqtt_bridge/scripts/mqtt_bridge_node.py", line 9, in <module>
    mqtt_bridge_node()
  File "/home/sami/mqtt/src/mqtt_bridge/src/mqtt_bridge/app.py", line 56, in mqtt_bridge_node
    mqtt_client.connect(**conn_params)
  File "/usr/local/lib/python2.7/dist-packages/paho/mqtt/client.py", line 768, in connect
    return self.reconnect()
  File "/usr/local/lib/python2.7/dist-packages/paho/mqtt/client.py", line 895, in reconnect
    sock = socket.create_connection((self._host, self._port), source_address=(self._bind_address, 0))
  File "/usr/lib/python2.7/socket.py", line 571, in create_connection
    raise err
socket.error: [Errno 111] Connection refused
[mqtt_bridge-1] process has died [pid 16425, exit code 1, cmd /home/sami/mqtt/src/mqtt_bridge/scripts/mqtt_bridge_node.py __name:=mqtt_bridge __log:=/home/sami/.ros/log/eb832402-17dc-11e8-a3ed-904ce5b24de1/mqtt_bridge-1.log].
log file: /home/sami/.ros/log/eb832402-17dc-11e8-a3ed-904ce5b24de1/mqtt_bridge-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
sami@sami-Vostro-1520:~/mqtt$

hesamira, 2018-05-17:
Hi,
I have faced with the following error.
Please, guide me.
Thanks.

sami@sami:~/mqtt_bridge$ roslaunch mqtt_bridge demo.launch
... logging to /home/sami/.ros/log/b8659f3a-59b5-11e8-9ff1-904ce5b24de1/roslaunch-sami-14646.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://sami:46410/

SUMMARY
========

PARAMETERS
 * /mqtt_bridge/bridge: [{'topic_from': '...
 * /mqtt_bridge/deserializer: msgpack:loads
 * /mqtt_bridge/mqtt/client/protocol: 4
 * /mqtt_bridge/mqtt/connection/host: localhost
 * /mqtt_bridge/mqtt/connection/keepalive: 60
 * /mqtt_bridge/mqtt/connection/port: 1883
 * /mqtt_bridge/mqtt/private_path: device/001
 * /mqtt_bridge/serializer: msgpack:dumps
 * /rosdistro: kinetic
 * /rosversion: 1.12.13

NODES
  /
    mqtt_bridge (mqtt_bridge/mqtt_bridge_node.py)

auto-starting new master
process[master]: started with pid [14662]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to b8659f3a-59b5-11e8-9ff1-904ce5b24de1
process[rosout-1]: started with pid [14676]
started core service [/rosout]
running rosparam delete /mqtt_bridge/
ERROR: parameter [/mqtt_bridge] is not set
process[mqtt_bridge-2]: started with pid [14698]
Traceback (most recent call last):
  File "/home/sami/mqtt_bridge/src/mqtt_bridge/scripts/mqtt_bridge_node.py", line 5, in <module>
    from mqtt_bridge.app import mqtt_bridge_node
  File "/home/sami/mqtt_bridge/src/mqtt_bridge/src/mqtt_bridge/app.py", line 8, in <module>
    from .bridge import create_bridge
  File "/home/sami/mqtt_bridge/src/mqtt_bridge/src/mqtt_bridge/bridge.py", line 10, in <module>
    from .util import lookup_object, extract_values, populate_instance
  File "/home/sami/mqtt_bridge/src/mqtt_bridge/src/mqtt_bridge/util.py", line 5, in <module>
    from rosbridge_library.internal import message_conversion
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosbridge_library/internal/message_conversion.py", line 45, in <module>
    from rosbridge_library.util import string_types, bson
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosbridge_library/util/__init__.py", line 24, in <module>
    "BSON installation does not support all necessary features. "
Exception: BSON installation does not support all necessary features. Please use the MongoDB BSON implementation. See: https://github.com/RobotWebTools/rosbridge_suite/issues/198
**[mqtt_bridge-2] process has died [pid 14698, exit code 1, cmd /home/sami/mqtt_bridge/src/mqtt_bridge/scripts/mqtt_bridge_node.py __name:=mqtt_bridge __log:=/home/sami/.ros/log/b8659f3a-59b5-11e8-9ff1-904ce5b24de1/mqtt_bridge-2.log].
log file: /home/sami/.ros/log/b8659f3a-59b5-11e8-9ff1-904ce5b24de1/mqtt_bridge-2*.log**

ninadkale98, 2018-06-11:
started roslaunch server http://ninad-VPCEH28FA:38903/

SUMMARY
========

PARAMETERS
 * /mqtt_bridge/+ deserializer: json:loads
 * /mqtt_bridge/+ serializer: json:dumps
 * /mqtt_bridge/-deserializer: msgpack:loads
 * /mqtt_bridge/-serializer: msgpack:dumps
 * /mqtt_bridge/bridge: [{'topic_from': '...
 * /mqtt_bridge/mqtt/client/protocol: 4
 * /mqtt_bridge/mqtt/connection/host: 127.0.0.1
 * /mqtt_bridge/mqtt/connection/keepalive: 60
 * /mqtt_bridge/mqtt/connection/port: 1883
 * /mqtt_bridge/mqtt/private_path: device/001
 * /rosdistro: kinetic
 * /rosversion: 1.12.13

NODES
  /
    mqtt_bridge (mqtt_bridge/mqtt_bridge_node.py)

auto-starting new master
process[master]: started with pid [6728]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 6ccb0c88-6d63-11e8-bc37-78843cee63d3
process[rosout-1]: started with pid [6741]
started core service [/rosout]
running rosparam delete /mqtt_bridge/
ERROR: parameter [/mqtt_bridge] is not set
process[mqtt_bridge-2]: started with pid [6763]
[INFO] [1528713444.238313]: MQTT connected

i am getting parametera are not set what to do now.my ping node is not able to talk with pong node.
mqtt is working fine when i am running it seperately .
thanks

welkerCode, 2018-08-01:
@ninadkale98 Did you ever figure out the parameter [/mqtt_bridge] is not set problem?  I get that message along with an ImportError: No module named rosbridge_library.internal in util.py line 5

mjforanVT, 2019-04-06:
> @ninadkale98 Did you ever figure out the parameter [/mqtt_bridge] is not set problem? I get that message along with an ImportError: No module named rosbridge_library.internal in util.py line 5

For "parameter not set," see [my comment on the other thread](https://github.com/groove-x/mqtt_bridge/issues/11#issuecomment-480520615). For the import error, you have to install rosbridge

dtmdyc, 2019-04-30:
sudo apt-get install ros-kinetic-rosbridge-suite

ledmonster, 2020-12-14:
duplicated with #11

closed 2020-12-14 by ledmonster