groove-x/mqtt_bridge #11: Error in: roslaunch mqtt_bridge demo.launch

jabobian, 2018-03-19:
When test mqtt_bridge, it says:  ERROR: parameter [/mqtt_bridge] is not set. Please help.

xx@ubuntu:~$ roslaunch mqtt_bridge demo.launch
... logging to /home/xx/.ros/log/17af3876-2b3e-11e8-ba85-08002773ca37/roslaunch-ubuntu-9564.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ubuntu:34693/

SUMMARY
========

PARAMETERS
 * /mqtt_bridge/bridge: [{'topic_from': '...
 * /mqtt_bridge/deserializer: msgpack:loads
 * /mqtt_bridge/mqtt/client/protocol: 4
 * /mqtt_bridge/mqtt/connection/host: localhost
 * /mqtt_bridge/mqtt/connection/keepalive: 60
 * /mqtt_bridge/mqtt/connection/port: 1883
 * /mqtt_bridge/mqtt/private_path: device/001
 * /mqtt_bridge/serializer: msgpack:dumps
 * /rosdistro: indigo
 * /rosversion: 1.11.21

NODES
  /
    mqtt_bridge (mqtt_bridge/mqtt_bridge_node.py)

auto-starting new master
process[master]: started with pid [9575]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 17af3876-2b3e-11e8-ba85-08002773ca37
process[rosout-1]: started with pid [9588]
started core service [/rosout]
running rosparam delete /mqtt_bridge/
ERROR: parameter [/mqtt_bridge] is not set
process[mqtt_bridge-2]: started with pid [9610]
[Errno 32] Broken pipe
[INFO] [WallTime: 1521440634.096355] MQTT disconnected
[INFO] [WallTime: 1521440635.110383] MQTT disconnected
[INFO] [WallTime: 1521440637.112733] MQTT disconnected
[INFO] [WallTime: 1521440641.116993] MQTT disconnected
[INFO] [WallTime: 1521440649.120404] MQTT disconnected

jabobian, 2018-03-19:
roslaunch-ubuntu-9564.log

[roslaunch][INFO] 2018-03-19 14:23:52,122: Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
[roslaunch][INFO] 2018-03-19 14:23:52,126: Done checking log file disk usage. Usage is <1GB.
[roslaunch][INFO] 2018-03-19 14:23:52,127: roslaunch starting with args ['/opt/ros/indigo/bin/roslaunch', 'mqtt_bridge', 'demo.launch']
[roslaunch][INFO] 2018-03-19 14:23:52,127: roslaunch env is {'SHELL': '/bin/bash', 'XDG_DATA_DIRS': '/usr/share/ubuntu:/usr/share/gnome:/usr/local/share/:/usr/share/', 'MANDATORY_PATH': '/usr/share/gconf/ubuntu.mandatory.path', 'COMPIZ_CONFIG_PROFILE': 'ubuntu', 'JOB': 'dbus', 'SESSION': 'ubuntu', 'XMODIFIERS': '@im=ibus', 'JAVA_HOME': '/opt/jdk1.8.0_25', 'TURTLEBOT_INTERACTIONS_LIST': '[turtlebot_bringup/admin.interactions, turtlebot_bringup/documentation.interactions, turtlebot_bringup/pairing.interactions, turtlebot_bringup/visualisation.interactions]', 'SELINUX_INIT': 'YES', 'XDG_RUNTIME_DIR': '/run/user/1000', 'PYTHONPATH': '/home/exbot/cartographer_ws/install_isolated/lib/python2.7/dist-packages:/opt/ros/indigo/lib/python2.7/dist-packages', 'COMPIZ_BIN_PATH': '/usr/bin/', 'ROS_ROOT': '/opt/ros/indigo/share/ros', 'XDG_SESSION_ID': 'c2', 'DBUS_SESSION_BUS_ADDRESS': 'unix:abstract=/tmp/dbus-XraJhnKVtZ', 'DEFAULTS_PATH': '/usr/share/gconf/ubuntu.default.path', 'DESKTOP_SESSION': 'ubuntu', 'GTK_MODULES': 'overlay-scrollbar:unity-gtk-module', 'INSTANCE': '', 'TURTLEBOT_3D_SENSOR': 'asus_xtion_pro', 'TURTLEBOT_RAPP_PACKAGE_BLACKLIST': '[]', 'XDG_MENU_PREFIX': 'gnome-', 'LS_COLORS': 'rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arj=01;31:*.taz=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.zip=01;31:*.z=01;31:*.Z=01;31:*.dz=01;31:*.gz=01;31:*.lz=01;31:*.xz=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.jpg=01;35:*.jpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.axv=01;35:*.anx=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.axa=00;36:*.oga=00;36:*.spx=00;36:*.xspf=00;36:', 'TURTLEBOT_TYPE': 'turtlebot', 'GNOME_DESKTOP_SESSION_ID': 'this-is-deprecated', 'XDG_CURRENT_DESKTOP': 'Unity', 'CPATH': '/home/exbot/cartographer_ws/install_isolated/include:/opt/ros/indigo/include:/opt/eigen:/opt/eigen', 'USER': 'exbot', 'XDG_VTNR': '7', 'CMAKE_PREFIX_PATH': '/home/exbot/cartographer_ws/install_isolated:/opt/ros/indigo', 'XAUTHORITY': '/home/exbot/.Xauthority', 'LANGUAGE': 'zh_CN:zh', 'SESSION_MANAGER': 'local/ubuntu:@/tmp/.ICE-unix/1866,unix/ubuntu:/tmp/.ICE-unix/1866', 'SHLVL': '1', 'QT_QPA_PLATFORMTHEME': 'appmenu-qt5', 'DISPLAY': ':0', 'CLUTTER_IM_MODULE': 'xim', 'WINDOWID': '65011723', 'ROS_ETC_DIR': '/opt/ros/indigo/etc/ros', 'GPG_AGENT_INFO': '/run/user/1000/keyring-To1Vpb/gpg:0:1', 'LUA_CPATH': ';;/opt/ros/indigo/lib/lua/5.1/?.so;/opt/ros/indigo/lib/lua/5.1/?.so', 'CLASSPATH': '/opt/jdk1.8.0_25/lib', 'GDMSESSION': 'ubuntu', 'PKG_CONFIG_PATH': '/home/exbot/cartographer_ws/install_isolated/lib/pkgconfig:/opt/ros/indigo/lib/pkgconfig:/opt/ros/indigo/lib/i386-linux-gnu/pkgconfig', 'ROS_PACKAGE_PATH': '/home/exbot/cartographer_ws/install_isolated/share:/home/exbot/cartographer_ws/install_isolated/stacks:/opt/ros/indigo/share:/opt/ros/indigo/stacks:/home/exbot/rosbuild_ws:/home/exbot/mqtt_bridge', 'TURTLEBOT_BASE': 'kobuki', '_': '/opt/ros/indigo/bin/roslaunch', 'GTK_IM_MODULE': 'ibus', 'XDG_CONFIG_DIRS': '/etc/xdg/xdg-ubuntu:/usr/share/upstart/xdg:/etc/xdg', 'TURTLEBOT_BATTERY': '/sys/class/power_supply/BAT0', 'ROS_MASTER_URI': 'http://localhost:11311', 'COLORTERM': 'gnome-terminal', 'XDG_GREETER_DATA_DIR': '/var/lib/lightdm-data/exbot', 'QT4_IM_MODULE': 'xim', 'XDG_SEAT_PATH': '/org/freedesktop/DisplayManager/Seat0', 'HOME': '/home/exbot', 'LD_LIBRARY_PATH': '/opt/ros/indigo/lib/typelib:/home/exbot/cartographer_ws/install_isolated/lib:/opt/ros/indigo/lib:/opt/ros/indigo/lib/i386-linux-gnu:/opt/ros/indigo/lib/typelib', 'LANG': 'zh_CN.UTF-8', 'ROS_LOG_FILENAME': '/home/exbot/.ros/log/17af3876-2b3e-11e8-ba85-08002773ca37/roslaunch-ubuntu-9564.log', 'TURTLEBOT_SIMULATION': 'false', 'TEXTDOMAIN': 'im-config', 'LUA_PATH': ';;/opt/ros/indigo/share/lua/5.1/?.lua;/opt/ros/indigo/share/lua/5.1/kdl_typekit/?.lua;/opt/ros/indigo/share/lua/5.1/?.lua;/opt/ros/indigo/share/lua/5.1/kdl_typekit/?.lua', 'VTE_VERSION': '3409', 'TURTLEBOT_SERIAL_PORT': '/dev/kobuki', 'LESSOPEN': '| /usr/bin/lesspipe %s', 'TURTLEBOT_NAME': 'turtlebot', 'LESSCLOSE': '/usr/bin/lesspipe %s %s', 'GNOME_KEYRING_PID': '1700', 'TYPELIB_USE_GCCXML': '1', 'RTCTREE_NAMESERVERS': 'localhost:15005', 'ROS_DISTRO': 'indigo', 'RUBYLIB': '/opt/ros/indigo/lib/ruby/1.9.1/i686-linux:/opt/ros/in [... cut ...]
[roslaunch][INFO] 2018-03-19 14:23:53,578: process[mqtt_bridge-2]: started with pid [9610]
[roslaunch][INFO] 2018-03-19 14:23:53,578: ... successfully launched [mqtt_bridge-2]
[roslaunch][INFO] 2018-03-19 14:23:53,579: ... launch_nodes complete
[roslaunch.pmon][INFO] 2018-03-19 14:23:53,579: registrations completed <ProcessMonitor(ProcessMonitor-1, started daemon -1233806528)>
[roslaunch.parent][INFO] 2018-03-19 14:23:53,579: ... roslaunch parent running, waiting for process exit
[roslaunch][INFO] 2018-03-19 14:23:53,579: spin

mjforanVT, 2019-04-06:
For the "parameter not set" issue, I was able to resolve this by editing the demo.launch file. There is a line "<rosparam command="delete" param="" />"  - I think this was deleting the bridge parameter, which was causing that error. I removed the line and the demo worked.

PR #44 (2021-01-01): don't delete params in demo.launch
close #11
closed 2021-01-01 by ledmonster