groove-x/mqtt_bridge #17: MQTT to ROS and get ERROR info unpack(b)  received extra data.

acdart, 2018-12-10:
When i try to pub mqtt message and transfer it to ros message

grejj, 2019-09-27:
Adding a '/' in front of the topic name in the mqtt pub command solved the issue for me.

PrimeshShamilka, 2020-05-11:
`mosquitto_pub -t 'echo' -m '{"data" : "hello"}'`

This solved the issue for me.

cihankurt98, 2020-10-15:
I am still having this issue. As for @grejj you are publishing directly to a ROS topic. The "/" is not meant for the MQTT topic I think.
As for @PrimeshShamilka your method still causes the error.

grejj, 2020-10-15:
Can you post your params.yaml file and the command you are using to publish mqtt message to ROS?

cihankurt98, 2020-10-15:
@grejj I could solve it by removing the msgpack serialization and just switching to json formatting. I think I needed to publish a msgpack formatted message to the mqtt topic in order to send it to ROS. By switching to json I can just simply publish a json format.

However I would still like to use msgpack, since it seems to be faster.

This is my yaml file:
```
mqtt:
  client:
    protocol: 4      # MQTTv311
  connection:
    host: 192.168.61.1
    port: 1883
    keepalive: 60
  private_path: device/001
serializer: json:dumps
deserializer: json:loads
bridge:
  # ping pong
  - factory: mqtt_bridge.bridge:RosToMqttBridge
    msg_type: std_msgs.msg:Bool
    topic_from: /ping
    topic_to: ping
  - factory: mqtt_bridge.bridge:MqttToRosBridge
    msg_type: std_msgs.msg:Bool
    topic_from: ping
    topic_to: /pong
  # echo back
  - factory: mqtt_bridge.bridge:RosToMqttBridge
    msg_type: std_msgs.msg:String
    topic_from: /echo
    topic_to: echo
  - factory: mqtt_bridge.bridge:MqttToRosBridge
    msg_type: std_msgs.msg:String
    topic_from: echo
    topic_to: /back
  # private path
  - factory: mqtt_bridge.bridge:RosToMqttBridge
    msg_type: std_msgs.msg:String
    topic_from: /private/echo
    topic_to: ~/echo
  - factory: mqtt_bridge.bridge:MqttToRosBridge
    msg_type: std_msgs.msg:String
    topic_from: ~/echo
    topic_to: /private/back
  # Velocity
  - factory: mqtt_bridge.bridge:RosToMqttBridge
    msg_type: std_msgs.msg:String
    topic_from: /velocity
    topic_to: DCMotorLogic/oPreferredVelocity
```
And these were the messages I tried to publish:
```
mosquitto_pub -h 192.168.61.1 -t 'ping' -m {"data: true"}
mosquitto_pub -h 192.168.61.1 -t 'ping' -m {"data": "true"}
```
I tried more formats, but somehow couldn't get the right format for msgpack.

grejj, 2020-10-15:
The mqtt message you are publishing has to be one long string I believe so I think you have to surround the message in single quotes.

Also the ping topic is of type bool so you shouldn't put " " around true or else it will think it is a string.

Try this:

$ mosquitto_pub -h 192.168.61.1 -t 'ping' -m '{"data": true}'

cihankurt98, 2020-10-15:
> The mqtt message you are publishing has to be one long string I believe so I think you have to surround the message in single quotes.
>
> Also the ping topic is of type bool so you shouldn't put " " [...]

That still gives me an error. 
`unpack(b) received extra data.
`
Atleast, when I try it with msgpack serialization. With Json serialization it works

grejj, 2020-10-15:
I am no sure how else I can help. I copied your params.yaml file and tried both json and msgpack serialization with this command:

```
$ mosquitto_pub -t 'ping' -m '{"data": true}'
```
And I was able to successfully get the true message when echo:
```
$  rostopic echo /pong 
```

The only other thing I can think of was that bson package was giving me problems and I am using pymongo instead, but shouldn't affect you since you said that json is working fine. These are the packages I have installed:

Inject==3.5.4
pymongo==3.11.0
paho-mqtt==1.3.1
msgpack-python==0.5.6

cihankurt98, 2020-10-15:
Strange. It appeared that I didn't have bson installed at all. As soon as I installed it, it said
`BSON installation does not support all necessary features.`

So I uninstalled it and installed pymongo. However that still returns the same error
`unpack(b) received extra data`

Is there a file where I have to replace bson with pymongo or is just installing enough?

PrimeshShamilka, 2020-10-16:
> I am still having this issue. As for @grejj you are publishing directly to a ROS topic. The "/" is not meant for the MQTT topic I think.
> As for @PrimeshShamilka your method still causes the error.

``` mosquitto_pub -t 'echo' -m '{"data" : "hello"}' ```

This code snippet only works with JSON as the serializer / deserializer.

In order to use msgpack, first you should  packing data into msgpack format before publishing from the mqtt publisher. This can be easily done in python as follows, 

```
msg = bytearray(msgpack.dumps({'data': "Hello"}))
 
# or 
 
msg = bytearray(msgpack.packb({'data': "True"}, use_bin_type=True))

```
When sending from ros to mqtt using mspack, follow the steps below,

- Publish from ros

``` rostopic pub /echo std_msgs/String "data: 'hello'" ```

- Unpack the msgpack message at the mqtt subscriber.

``` unpacked_msg = msgpack.unpackb(msg.payload, use_list=False, raw=False) ```

Hope this will help.

ledmonster, 2021-01-01:
Please refer to #34

closed 2021-01-01 by ledmonster

aamirhatim, 2023-03-21:
Opening this up again since I'm getting the same error, but I have a different scenario:

Machine 1:
- Jetson Nano (Ubuntu 18.04)
- ROS Melodic
- Python 2.7

Machine 2:
- Laptop (Ubuntu 20.04)
- ROS Noetic
- Python 3

I am trying to send a PointCloud2 message from Machine 1 to Machine 2. Whenever Machine 2 gets a message, I get the following error message in my terminal: `unpack(b) received extra data.`
I'm curious if there is a breaking change with `msgpack-python` when going from Python version 2 to version 3. The [msgpack project page](https://pypi.org/project/msgpack/) says the following:

---

**Major breaking changes in msgpack 1.0**
- Python 2
    - The extension module does not support Python 2 anymore. The pure Python implementation (`msgpack.fallback`) is used for Python 2.
- Packer
    - `use_bin_type=True` by default. bytes are encoded in bin type in msgpack. **If you are still using Python 2, you must use unicode for all string types.** You can use `use_bin_type=False` to encode into old msgpack format.
    - `encoding` option is removed. UTF-8 is used always.
- Unpacker
    - `raw=False` by default. It assumes str types are valid UTF-8 string and decode them to Python str (unicode) object.
    - `encoding` option is removed. You can use `raw=True` to support old format.
    - Default value of `max_buffer_size` is changed from 0 to 100 MiB.
    - Default value of `strict_map_key` is changed to True to avoid hashdos. You need to pass `strict_map_key=False` if you have data which contain map keys which type is not bytes or str.

---

Using JSON as the serializer/deserializer works fine, however. I also don't see any msgpack errors when using the bridge between two machines using ROS Noetic.

Here are my config files:

Machine 1:
```
mqtt:
    client:
        protocol: 4      # MQTTv311
    connection:
        host: 192.168.20.3
        port: 1883
        keepalive: 60
bridge:
    - factory: mqtt_bridge.bridge:RosToMqttBridge
      msg_type: sensor_msgs.msg:PointCloud2
      topic_from: /jackal01/velodyne_points
      topic_to: jackal01/velodyne_points

#serializer: json:dumps
#deserializer: json:loads
```

Machine 2:
```
mqtt:
    client:
        protocol: 4      # MQTTv311
    connection:
        host: 192.168.20.3
        port: 1883
        keepalive: 60
bridge:
    - factory: mqtt_bridge.bridge:MqttToRosBridge
      msg_type: sensor_msgs.msg:PointCloud2
      topic_from: /jackal01/velodyne_points
      topic_to: /jackal01/velodyne_points

#serializer: json:dumps
#deserializer: json:loads
```