groove-x/mqtt_bridge #30: Unable to install inject

mazharshaikh0593, 2020-08-21:
Hello,

I'm trying to use this package for my Master's project work. 

I have Ubuntu 20.4 with ROS noetic python3 as well as Ubuntu 16 with ROS kinetic python 2.7.

On noetic, I was able to install all the libraries and run mqtt_bridge successfully, but the codes in app.py and bridge.py don't support python3 language. For example, it doesn't take the base string data type. I tried using str instead-but in line 77 of bridge.py its causing me encoding errors. Do you have any workaround for this? Or m I missing something?

On kinetic, I m unable to install the inject python module. I ran pip2 as well as pip with both suggestions given in this forum.
Error: Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-JEKfkg/inject/
It seems the inject python module doesn't support python2 now and only python3 is working with it.

Any suggestions or help is welcome. I tried lots of things. Thank you in advance.

darkasassin44, 2020-08-31:
Same problem.
I'm using Parallels to run Ubuntu 18.04(ROS melodic) on my Mac.
Unfortunately, it's my first time I use mqtt_birdge.
While I follow the guideline, installing requirements.txt, the same error occurred.
[image]

So when I try to run the launch file, the following error has occurred.
[image]

When you could run this mqtt_bridge successfully on Ubuntu 16, haven't you seen this error?

darkasassin44, 2020-08-31:
I found the solution for yours and mine.
https://pypi.org/project/Inject/#description

check here to confirm both your python and inject module version.
for me, I built ROS packages with python 2.7.
[image]
therefore, I should have installed inject version 3.x.

so I did
`pip install Inject==3.5.4`
and it solved.

mazharshaikh0593, 2020-08-31:
I was able to workaround that issue by manually installing the dependencies
rather than through txt file.

Thank you.

mengkeroborock, 2020-11-17:
他

> I found the solution for yours and mine.
> https://pypi.org/project/Inject/#description
>
> check here to confirm both your python and inject module version.
> for me, I built ROS packages with python 2.7. [...]

good ～！

ledmonster, 2020-12-13:
Sorry for replying too late. I fixed this issue on #37.

closed 2020-12-13 by ledmonster