groove-x/mqtt_bridge #31: roslaunch Error

darkasassin44, 2020-08-31:
I have checked reqiurements.txt successfully, and only remained launch the `demo.launch` file.

```
conceptbug95@ubuntu:~$ roslaunch mqtt_bridge demo.launch
... logging to /home/conceptbug95/.ros/log/f08b33d8-eb5e-11ea-bbdb-001c42b01667/roslaunch-ubuntu-32640.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://localhost:36857/

SUMMARY
========

PARAMETERS
 * /mqtt_bridge/bridge: [{'topic_from': '...
 * /mqtt_bridge/deserializer: msgpack:loads
 * /mqtt_bridge/mqtt/client/protocol: 4
 * /mqtt_bridge/mqtt/connection/host: 10.0.0.254
 * /mqtt_bridge/mqtt/connection/keepalive: 60
 * /mqtt_bridge/mqtt/connection/port: 1883
 * /mqtt_bridge/serializer: msgpack:dumps
 * /rosdistro: melodic
 * /rosversion: 1.14.9

NODES
  /
    mqtt_bridge (mqtt_bridge/mqtt_bridge_node.py)

ROS_MASTER_URI=http://localhost:11311

running rosparam delete /mqtt_bridge/
process[mqtt_bridge-1]: started with pid [32660]
Traceback (most recent call last):
  File "/home/conceptbug95/catkin_ws/src/mqtt_bridge/scripts/mqtt_bridge_node.py", line 5, in <module>
    from mqtt_bridge.app import mqtt_bridge_node
  File "/home/conceptbug95/catkin_ws/src/mqtt_bridge/src/mqtt_bridge/app.py", line 8, in <module>
    from .bridge import create_bridge
  File "/home/conceptbug95/catkin_ws/src/mqtt_bridge/src/mqtt_bridge/bridge.py", line 10, in <module>
    from .util import lookup_object, extract_values, populate_instance
  File "/home/conceptbug95/catkin_ws/src/mqtt_bridge/src/mqtt_bridge/util.py", line 5, in <module>
    from rosbridge_library.internal import message_conversion
  File "/home/conceptbug95/catkin_ws/src/rosbridge_suite/rosbridge_library/src/rosbridge_library/internal/message_conversion.py", line 45, in <module>
    from rosbridge_library.util import string_types, bson
  File "/home/conceptbug95/catkin_ws/src/rosbridge_suite/rosbridge_library/src/rosbridge_library/util/__init__.py", line 24, in <module>
    "BSON installation does not support all necessary features. "
Exception: BSON installation does not support all necessary features. Please use the MongoDB BSON implementation. See: https://github.com/RobotWebTools/rosbridge_suite/issues/198
[mqtt_bridge-1] process has died [pid 32660, exit code 1, cmd /home/conceptbug95/catkin_ws/src/mqtt_bridge/scripts/mqtt_bridge_node.py __name:=mqtt_bridge __log:=/home/conceptbug95/.ros/log/f08b33d8-eb5e-11ea-bbdb-001c42b01667/mqtt_bridge-1.log].
log file: /home/conceptbug95/.ros/log/f08b33d8-eb5e-11ea-bbdb-001c42b01667/mqtt_bridge-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

But I don't know how to fix this error.
Even there are no error codes or error messages.

PLEASE guide me how to solve this problem if you know how to do.

asoriano1, 2020-09-03:
do this:

sudo pip uninstall bson
python -m pip install pymongo

darkasassin44, 2020-09-03:
I'm a beginner with Ubuntu, so is there any difference between 
`python -m pip install pymongo` and `pip install pymongo`?
I solved this problem yesterday just before I left my office, but cannot understand your suggestion.
Of course, your suggestion also works too! :)

closed 2020-11-02 by darkasassin44

joko06, 2022-02-23:
@darkasassin44 How did you solved this problem?