groove-x/mqtt_bridge #34: Keep getting "Data" in message

cihankurt98, 2020-10-15:
Let's say im sending the following message:

`rostopic pub /velocity std_msgs/String "hello"`

I get the following in my MQTT client:
`��data�hello`

As you can see, I never said something about the unknown character and data part.
Where is it coming from? From messagepack? How can I remove it?

cihankurt98, 2020-10-15:
I could remove the strange characters by switching from msgpack to json serialization. However, I don't want the "data" part. I can't do a string remove aswell, somehow it doesn't recognize it.

cihankurt98, 2020-10-16:
It seems like when I go from MQTT to ROS I always get a deserialization error. No matter what serialization package I use. I tried msgpack and cbor2. There is always more data expected then there actually is. It happens when the mqtt_msg is being converted to a ROS message.

mayershoc, 2020-11-07:
Hi cihankurt98, 
are you using python 2 or 3? If you are using python3, this might be related to a deserialization issue I used to get. Feel free to give my [fork](https://github.com/mayershoc/mqtt_bridge/tree/feature/python3) a try.

ledmonster, 2021-01-01:
Sorry for replying too late. This is a specification. ROS primitive data type like `Bool` or `String` has its message on `data` attribute. 

- http://docs.ros.org/en/melodic/api/std_msgs/html/msg/String.html
- http://docs.ros.org/en/melodic/api/std_msgs/html/msg/Bool.html

So these would be serialized to like `{"data": True}`.

closed 2021-01-01 by ledmonster