groove-x/mqtt_bridge #35: python3 compatibility (basestring)

Otacon5555, 2020-11-03:
Had to substitute `basestring` with `str` in `app.py` and `bridge.py` to make it work with python3.
I could push the commit if you let me.
Thanks.

Otacon5555, 2020-11-03:
Don't know if it is related, but running the demo (after having removed a line from demo.launch like [explained here](https://github.com/groove-x/mqtt_bridge/issues/11#issuecomment-480520615), i get the following error:

[INFO] [1604420450.125915]: MQTT connected
[ERROR] [1604420468.085236]: sequence item 0: expected str instance, bytes found

Any ideas?

mayershoc, 2020-11-07:
Hey Otacon5555, 
I had the same issue and fixed it in my [fork](https://github.com/mayershoc/mqtt_bridge/tree/feature/python3). 
Give it a try and let me know if you come across any problems.
BR, Chris

Otacon5555, 2020-11-09:
Thank you very much mayershoc,
your fork is working fine for me!

I'll close this issue then and use your version.
Thank again!

closed 2020-11-09 by Otacon5555