groove-x/mqtt_bridge #42: How to configure will params in demo.yaml

baycxh, 2020-12-24:
How to configure will params in demo.yaml , is there any params to be configure and what's the pattern?
Thanks!

ledmonster, 2021-01-01:
Hi @baycxh.
Which params do you want to configure? If you run mosquitto in local, you can use demo.yaml as it is.

baycxh, 2021-01-02:
@ledmonster Thanks for your answer.  I want to know how to set the will params in the configure to ensure the last message from client is sent.

ledmonster, 2021-01-02:
Oh, I see. I haven't tried, but this may work.

```yaml
mqtt:
  client:
    protocol: 4      # MQTTv311
  connection:
    host: localhost
    port: 1883
    keepalive: 60
  private_path: device/001
  will:
    topic: some_topic
    payload: some_payload
    qos: 0
    retain: false
    # properties: # we don't support this option, since it needs to be a Properties instance.
```

Related code:
- https://github.com/groove-x/mqtt_bridge/blob/master/src/mqtt_bridge/mqtt_client.py#L43-L45
- https://github.com/eclipse/paho.mqtt.python/blob/master/src/paho/mqtt/client.py#L1679

abhishek47kashyap, 2021-05-14:
@ledmonster 
Thanks for posting the links. I'm trying to do the same thing ("_ensure the last message from client is sent_") and needed to better understand how to set up `will` params: 

From the `will_set()` method in the second link:

> **topic**: The topic that the will message should be published on.
> 
> **payload**: The message to send as a will. If not given, or set to None a
>          zero length message will be used as the will. Passing an int or float
>          will result in the payload being converted to a string representing
>          that number. If you wish to send a true int/float, use struct.pack() to
>          create the payload you require.

In the example below I want to transmit string messages to a couple of receivers:
```
bridge:
  # receiver 1
  - factory: mqtt_bridge.bridge:RosToMqttBridge
    msg_type: std_msgs.msg:String   
    topic_from: /to_receiver_1
    topic_to: to_receiver_1
  # receiver 2
  - factory: mqtt_bridge.bridge:RosToMqttBridge
    msg_type: std_msgs.msg:String   
    topic_from: /to_receiver_2
    topic_to: to_receiver_1
```
### My questions:
(I'm trying to achieve the same as publisher-latching in ROS)
1. If I wanted to set `retain: true` for both receivers, how/where would I insert `will` params?
2. How does `topic` relate to `topic_from` and `topic_to`?
3. Wouldn't `payload` be changing as messages are published on a topic?

closed 2021-10-09 by ledmonster