groove-x/mqtt_bridge #49: Adding header meta info to each message

pavloblindnology, 2021-03-04:
Hi,
It's quite a generic option to add some meta info to each message, like timestamp, uuid, etc.
Can be used for logging, latency measurements, etc.
I'm considering adding such functionality and contributing it.
What is the preferred way to do this?

1. Optionally add timestamp & uuid fields to the outcoming message, and strip them from incoming messages
```
{
  timestamp: XXXX
  uuid: YYYY
  ...
  // ROS message fields
  ...
}
```
2. Optionally move ROS outcoming message body to some MQTT message `payload` field and add timestamp & uuid fields to MQTT message `header` field, and vice verse for incoming messages
```
{
  header: {
    timestamp: XXXX
    uuid: YYYY
  }
  payload: {
    ...
    // ROS message fields
    ...
  }
}
```

3. Add another generic Bridge plugin implementing option 1/2
4. Other

Thanks

ledmonster, 2021-10-09:
@pavloblindnology 
Hi, sorry for replying late. You can define your custom `serializer` and `deserializer`, and configure it as rosparam.

I'll close this issue, but if you have another opinion, please reopen it.

closed 2021-10-09 by ledmonster

pavloblindnology, 2021-10-11:
@ledmonster Ok. But I consider developing custom (de)serializer a quite more heavier work than just updating the bridges. In this case I prefer to duplicate existing bridges with extended functionality.