groove-x/mqtt_bridge #50: when i input roslaunch mqtt_bridge demo.launch

975150313, 2021-03-07:
When I run roslaunch mqtt_bridge demo.launch in docker, he prompts me：
RLException: [demo.launch] is neither a launch file in package [mqtt_bridge] nor is [mqtt_bridge] a launch file name
The traceback for the exception was written to the log file

The docker is  ros:noetic
and I setup  ros-noetic-rosbridge-suite，the roslaunch rosbridge_server rosbridge_websocket.launch is working
what should I do？

bindu030, 2021-09-30:
> The docker is ros:noetic and I setup ros-noetic-rosbridge-suite，the roslaunch rosbridge_server rosbridge_websocket.launch is working what should I do？
How you are able to ahceive this?

bindu030, 2021-09-30:
> When I run roslaunch mqtt_bridge demo.launch in docker, he prompts me： RLException: [demo.launch] is neither a launch file in package [mqtt_bridge] nor is [mqtt_bridge] a launch file name The [...]