groove-x/mqtt_bridge #55: ROS2 mqtt_bridge?

HappySamuel, 2021-05-12:
Hi Author

Your mqtt_bridge is nice to use. By the way, is there any plan to release a ROS2 mqtt_bridge?

Best,
Samuel

130s, 2021-06-29:
There's also a question https://answers.ros.org/question/369262/can-we-use-mqtt-in-ros2/

flynneva, 2021-06-29:
you could try out this new package called [rospy2](https://github.com/dheera/rospy2) that should just be a one-line change to port it to ROS2 i think....

HappySamuel, 2021-06-30:
Thanks guys, if i have time i will try to look into the current mqtt_bridge and see how we can migrate into ROS2 and submit a PR.

kapilPython, 2021-09-25:
Hi Repository maintainers @ledmonster @yuma-m I have begin the implementation of ROS2 support for mqtt_bridge. It would be great if you could create a new branch for ROS2 in the repository. Please let me know what are your plans on ROS2 support.

ledmonster, 2021-09-27:
@kapilPython Sorry, I don't maintain this package now. It may be better to fork this project. I'll frozen the project after that.

ledmonster, 2021-10-09:
@kapilPython 

> It would be great if you could create a new branch for ROS2 in the repository.

I can't work on this project now, but I created a `ros2` branch. I'd appreciate you if you would implement ros2 support. I'll review PRs when you'll make PRs on this branch. Thank you.

kapilPython, 2021-12-03:
Hi @ledmonster I was busy with some personal work and therefore could not reply to you. It makes more sense to keep the same repository for the ROS2 mqtt_bridge so that people can get access to the mqtt_bridge by switching to a relevant branch.
But, if you suggest a new repository that's also fine.
For communication with ROS2 the same code infrastructure used in ROS1 can be used, the only changes which are required are usage of time in ros2, node creation and publisher, subscriber creation in ROS2.
I have also observed that eProsima has not implemented an integration service https://github.com/eProsima/Integration-Service for mqtt and our extension for ROS2 can be a help for the community.

ledmonster, 2021-12-13:
If anyone can contribute to ros2 support, please create a PR to `ros2` branch. Thank you.

kapilPython, 2021-12-13:
Hi @ledmonster you could try the first version of ros2 compatible mqtt bridge on my forked repository:
https://github.com/kapilPython/mqtt_bridge/tree/ros2

There is still some documentation, cleaning up and creating launch file left. I will take care of the same over the weekend. 

For trying the code just install ros2 and requirement.txt 
Then build the code with `colcon build`
then, `source install/setup.bash`
and run `ros2 run mqtt_bridge mqtt_bridge_node --ros-args --params-file ~/mqtt_ws/src/mqtt_bridge/config/demo_params.yaml`

kapilPython, 2021-12-13:
> Hi @ledmonster you could try the first version of ros2 compatible mqtt bridge on my forked repository: https://github.com/kapilPython/mqtt_bridge/tree/ros2
>
> There is still some documentation, [...]

After clean up would make a pull request.

kapilPython, 2021-12-16:
https://github.com/groove-x/mqtt_bridge/pull/62 Pull request for the ROS2 feature. Lets close the issue soon :relaxed: :relieved:

ledmonster, 2022-04-25:
I don't have enough time to maintain this project, so if someone is willing to contribute to the project, I'm glad to transfer the ownership. Anyway, hopefully I'll merge #62 soon. Thank you @kapilPython .

kapilPython, 2022-06-28:
@ledmonster I guess this issue can be also closed now.

ledmonster, 2022-06-29:
ROS2 is now supported on ros2 branch. 
See: https://github.com/groove-x/mqtt_bridge/pull/62

closed 2022-06-29 by ledmonster