groove-x/mqtt_bridge #59: Encoding problem on Ubuntu20 Ros Noetic

iscsrl, 2021-09-02:
Hi,
I recently moved my project to Ubuntu 20.04 and ROS Noetic from 18.04 and Melodic.
I installed on the new environment the ros-noetic-mqtt-bridge executing the same exacly code, but I'm experiencing a very weird problem. It looks like the bridge is publishing, on my mosquitto mqtt broker, messages with a wrong encoding despite all other publications coming from the classic mosquitto client are correctly encoded.

I tried with no luck to add the serialize/deserialize params in config:

```
mqtt:
  serializer:
    json: dumps
  deserializer:
    json: loads
  client:
    protocol: 4      # MQTTv311
```

There are other parameters allowed on serializer that helps me specifying the published message encoding?

HappySamuel, 2021-09-22:
Hi

I am using mqtt_bridge with ROS Noetic on Ubuntu 20.04 as well. But i didn't encounter the issue you mentioned.

This is my config for your reference:
[image]

Hope it helps.

Best,
Samuel

ledmonster, 2021-10-09:
@iscsrl 
Umm, it's weird. Have you resolved the issue? If not, please reopen and provide extra information.

closed 2021-10-09 by ledmonster