groove-x/mqtt_bridge #61: About mqtt_ Bridge is linked for a long time and cannot receive data

Minpig, 2021-09-22:
A few days after the mqttbridge link, the ROS client cannot receive data from the mqtt server, but the device can receive instructions sent to the mqtt server. At this point, restart mqttbridge and you can receive it again. Why? Is there a link timeout operation in mqttbridge that does not receive data?

HappySamuel, 2021-09-22:
I am also facing similar issue. Whenever the mqtt showing **disconnected** and **connected**, the communication between ROS client and MQTT server no longer established. But if i re-do **roslaunch** of mqtt bridge, and it shows connected, then the communication is established.

Why would the **connected** after **disconnected** doesn't actually connect ?

Best,
Samuel

chengchenglee, 2021-09-25:
Hello @Minpig @HappySamuel 

I have the same problem here.

If I use local environment to do the test. 
everything is fine.

if I use aws iot mqtt,
the log is like below.
[image]

it is really annoying and I spend quite a long time to do the trouble shoot.
and it seems that in the file bridge.py.
_callback_mqtt function is not called.

@HappySamuel 
hello Samuel.
what do you mean by **re-do roslaunch**
how do you make it work?could you please share what you do?

chengchenglee, 2021-09-25:
additional information.
I could use mosquitto to receive the data from aws iot 8883 port.

[image]

@yuma-m @ledmonster 
お二人、何か心当たりがございますでしょうか？

chengchenglee, 2021-09-27:
for adding more information.
sometimes, the execution is good. 
I added some log printout 
but still not good with error
[image]

**_MQTT disconnected_** 
issue still comes

@ledmonster @yuma-m 
Could you please help?

chengchenglee, 2021-09-27:
[image]
the situation is that first
MQTT disconnected

then 
MQTT connected

I am using aws iot tls.
I use the same certificated on mqtt.fx to do the test ,everything is fine.
@yuma-m @ledmonster

Minpig, 2021-10-08:
@chengchenglee @HappySamuel 
My situation is different from yours. You have encountered the error "mqtt disconnected" before. This is because the client_id is repeated. If two devices use this ID at the same time, the other device will be disconnected.
My question is that mqtt_bridge is always "mqtt connect". I can send and receive data normally for several consecutive days. But I can't get the data from the topic "mqtt_bridge_to_ros" a week later. Why? But I can send and receive the data from the topic "ros_to_mqtt_bridge".

chengchenglee, 2021-10-09:
@Minpig thanks for your information!
for your question, I wonder whether the payload of the mqtt is too large.
I encountered the problem that if I tried to use mqtt to transmit image info. it failed.

chengchenglee, 2021-10-09:
> @chengchenglee @HappySamuel My situation is different from yours. You have encountered the error "mqtt disconnected" before. This is because the client_id is repeated. If two devices use this ID at [...]

@Minpig 
for re-connection issue, I use my local PC for ROS->mqtt and mqtt->ROS bridge.
also, I did not set client id in params file.

when I only connect ROS->mqtt bridge, everything goes well.
[image]

adversely, when I only connect mqtt->ROS bridge, re-connection appears.
[image]
 
do you have any idea about this symptom?

foreverbala, 2023-01-21:
Did anybody got solution for this? I am also stucked in this issue . My ros environment is connected to the outside server. But I can't receive my messages from the server. How can I get my messages from mqtt_topic to rostopic .

sugangandhi, 2023-01-21:
> I am also facing similar issue. Whenever the mqtt showing **disconnected** and **connected**, the communication between ROS client and MQTT server no longer established. But if i re-do **roslaunch** [...]

Did you solve the problem, now im facing the same issue.

BRoboticsSHADO, 2023-01-25:
Hi @sugangandhi 

I added a topic to be published up into mqtt and then loop back to receive it. With this, an additional program is used to monitor the topic to check the connectivity. If the subscribed topic is missing, then kill the mqtt and re-launch it. 

Best
Samuel

HappySamuel, 2023-01-25:
Hi @sugangandhi

I added a topic to be published up into mqtt and then loop back to receive it. With this, an additional program is used to monitor the topic to check the connectivity. If the subscribed topic is missing, then kill the mqtt and re-launch it.

Best
Samuel

sugangandhi, 2023-01-25:
> Hi @sugangandhi
> 
> I added a topic to be published up into mqtt and then loop back to receive it. With this, an additional program is used to monitor the topic to check the connectivity. If the subscribed topic is missing, then kill the mqtt and re-launch it.
> 
> Best Samuel

Could you please give me an idea on how you added a topic into mqtt?

foreverbala, 2023-01-25:
Hi @HappySamuel 
I am receiving the following error as a response from broker server.
[image]
Can you please tell how to solve this?

HappySamuel, 2023-01-31:
Hi @sugangandhi 

For example the config for mqtt bridge, you will nee

bridge:    
  # ROS to MQTT
  - factory: mqtt_bridge.bridge:RosToMqttBridge
    msg_type: rmf_fleet_msgs.msg:RobotState
    topic_from: /amr/robot_state
    topic_to: /mqtt/robot_state     **`must be the same as`**
   # MQTT to ROS
  - factory: mqtt_bridge.bridge:MqttToRosBridge
    msg_type: rmf_fleet_msgs.msg:RobotState
    topic_from: /mqtt/robot_state   **`must be the same as`**
    topic_to: /test/robot_state   `Listen to this topic to monitor if the mqtt connection is OK or not`

Best
Samuel

foreverbala, 2023-01-31:
Hi @HappySamuel 

Thanks for your last comment.
As you said I did and now I am receiving the data.I am using std_msgs as msg type.But it shows error when I use other custom messages.
How did you created this `msg_type: rmf_fleet_msgs.msg:RobotState`

Can you please explain how to publish the received messages from mqtt topic in a new rostopic .
In clear, Now I am receiving the messages from mqtt topic because I subscribed it.Now I want to publish those received messages in a new rostopic when publishing the rostopic using `rostopic echo 'topic name'`. For that how can I get the mqtt messages to the newly created rostopic.

Thanks in advance
Bala

HappySamuel, 2023-01-31:
Hi @foreverbala 

It's custom msg i created, you may follow tutorial on how to create custom message. Regardless of custom message, mqtt bridge also can work with default ROS1 message, ex: **std_msgs** / **sensor_msgs**

You don't have to publish the received messages from mqtt topic into rostopic, `mqtt_bridge` will do the job, you just need to study to use the config setting. For publishing those received messages in new rostopic, you can just edit the `topic_to:` Or put a topic relay in between, then it will subscribe current ros topic name and relay it as another topic name.

Best
Samuel

foreverbala, 2023-01-31:
Hi @HappySamuel 

Thanks for your valuable answers.It will help me a lot.