groove-x/mqtt_bridge #64: pip3 install -r requirements.txt could not open directory

mit6058, 2022-02-23:
[Errno 2] No such file or directory: 'requirements.txt'

sairajkamble, 2022-02-23:
The name of the requirements file is dev-requirements.txt. So give the command pip3 install -r dev-requirements.txt

plumewind, 2022-03-17:
hello ,

also ERROR:

Defaulting to user installation because normal site-packages is not writeable
ERROR: Could not open requirements file: [Errno 2] No such file or directory: 'dev-requirements.txt'

plumewind, 2022-03-17:
sorry, I got it
$ cd ~/catkin_ws/src/mqtt_bridge
$  pip3 install -r dev-requirements.txt

plumewind, 2022-03-17:
Hello，

I got a error like:

Collecting geometry-msgs
  WARNING: Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f939fa9a8d0>: Failed to establish a new connection: [Errno 111] Connection refused',)': /rospypi/simple/any/geometry-msgs/geometry_msgs-1.13.0.post2-py2.py3-none-any.whl
  WARNING: Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f939fa9ab38>: Failed to establish a new connection: [Errno 111] Connection refused',)': /rospypi/simple/any/geometry-msgs/geometry_msgs-1.13.0.post2-py2.py3-none-any.whl
  WARNING: Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f939fa9acc0>: Failed to establish a new connection: [Errno 111] Connection refused',)': /rospypi/simple/any/geometry-msgs/geometry_msgs-1.13.0.post2-py2.py3-none-any.whl
  WARNING: Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f939fa9ae48>: Failed to establish a new connection: [Errno 111] Connection refused',)': /rospypi/simple/any/geometry-msgs/geometry_msgs-1.13.0.post2-py2.py3-none-any.whl
  WARNING: Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f939fa9afd0>: Failed to establish a new connection: [Errno 111] Connection refused',)': /rospypi/simple/any/geometry-msgs/geometry_msgs-1.13.0.post2-py2.py3-none-any.whl
ERROR: Could not install packages due to an OSError: HTTPSConnectionPool(host='raw.githubusercontent.com', port=443): Max retries exceeded with url: /rospypi/simple/any/geometry-msgs/geometry_msgs-1.13.0.post2-py2.py3-none-any.whl (Caused by NewConnectionError('<pip._vendor.urllib3.connection.HTTPSConnection object at 0x7f939faa3198>: Failed to establish a new connection: [Errno 111] Connection refused',))

What should I do?